moveit/moveit_task_constructor #667: Constrain grasp pose generator angle

therealbeni, 2025-03-17:
I'm using the MTC application for my robot arm. How can I constrain the approach direction so that only grasp poses from the front +-5° are considered as viable options for the grasp position. I currently have found no way to not use the pose generator that computes the positions all around the object.

`{
      // Sample grasp pose
      auto stage = std::make_unique<mtc::stages::GenerateGraspPose>("generate grasp pose");
      stage->properties().configureInitFrom(mtc::Stage::PARENT);
      stage->properties().set("marker_ns", "grasp_pose");
      stage->setPreGraspPose("open");
      stage->setObject("object");
      stage->setAngleDelta(M_PI / 12);
      stage->setMonitoredStage(current_state_ptr);  // Hook into current state

      Eigen::Isometry3d grasp_frame_transform;
      Eigen::Quaterniond q = Eigen::AngleAxisd(M_PI / 2, Eigen::Vector3d::UnitX()) *
                            Eigen::AngleAxisd(M_PI / 2, Eigen::Vector3d::UnitY()) *
                            Eigen::AngleAxisd(M_PI / 2, Eigen::Vector3d::UnitZ());
      grasp_frame_transform.linear() = q.matrix();
      grasp_frame_transform.translation().z() = 0.1;

      // Compute IK
      auto wrapper =
      std::make_unique<mtc::stages::ComputeIK>("grasp pose IK", std::move(stage));
      wrapper->setMaxIKSolutions(8);
      wrapper->setMinSolutionDistance(1.0);
      wrapper->setIKFrame(grasp_frame_transform, hand_frame);
      wrapper->properties().configureInitFrom(mtc::Stage::PARENT, { "eef", "group" });
      wrapper->properties().configureInitFrom(mtc::Stage::INTERFACE, { "target_pose" });
      grasp->insert(std::move(wrapper));
    }`

Is there a property that lets me constrain the approach angle?

rhaschke, 2025-03-17:
You could either use a `PredicateFilter` wrapper to filter the generated solution or - even better - use `GenerateRandomPose` which allows more flexible specification of the distributions.

therealbeni, 2025-04-07:
I currently have tried different options. The generateGraspPose stage works consistently, but sometimes the approach angle isn't always the same. I've tried with GenerateRandomPose, but couldn't get it to work.

```cpp
{
        auto stage = std::make_unique<stages::GenerateGraspPose>("generate grasp pose");
        stage->properties().configureInitFrom(Stage::PARENT);
        stage->properties().set("marker_ns", "grasp_pose");
        stage->setPreGraspPose("open");
        stage->setObject(object.name);
        stage->setAngleDelta(M_PI / 16);
        stage->setMonitoredStage(current_state_ptr_); // Hook into current state

        Eigen::Isometry3d grasp_frame_transform;
        Eigen::Quaterniond q = Eigen::AngleAxisd(M_PI / 2, Eigen::Vector3d::UnitX()) *
                               Eigen::AngleAxisd(M_PI / 2, Eigen::Vector3d::UnitY()) *
                               Eigen::AngleAxisd(0, Eigen::Vector3d::UnitZ());
        grasp_frame_transform.linear() = q.matrix();
        grasp_frame_transform.translation().z() = 0.1;

        // Compute IK
        auto wrapper =
            std::make_unique<stages::ComputeIK>("grasp pose IK", std::move(stage));
        wrapper->setMaxIKSolutions(8);
        wrapper->setMinSolutionDistance(1.0);
        wrapper->setIKFrame(grasp_frame_transform, hand_frame_);
        wrapper->properties().configureInitFrom(Stage::PARENT, {"eef", "group"});
        wrapper->properties().configureInitFrom(Stage::INTERFACE, {"target_pose"});
        grasp->insert(std::move(wrapper));
    }

```

Another approach I tried was to use a FixedCartesianPose. This solves the angle problem, however the position referenced is now always the initial position of the object. It doesnt references the object itself as in the GenerateGraspPose stage:

```cpp
{
        auto stage = std::make_unique<stages::FixedCartesianPoses>("generate grasp pose");
        stage->properties().configureInitFrom(Stage::PARENT);
        stage->properties().set("marker_ns", "grasp_pose");
        stage->setMonitoredStage(current_state_ptr_); // Hook into current state

        // --- Position ---
        Eigen::Isometry3d grasp_pose = Eigen::Isometry3d::Identity();
        grasp_pose.translation().x() = object.pose.pose.position.x;
        grasp_pose.translation().y() = object.pose.pose.position.y;
        grasp_pose.translation().z() = object.pose.pose.position.z;

        // --- Orientation from approach_direction ---
        const auto &approach = approach_direction;
        Eigen::Vector3d z_axis(approach.vector.x, approach.vector.y, approach.vector.z);
        z_axis.normalize();

        // Use fixed x-axis
        Eigen::Vector3d x_axis(1, 0, 0);
        if (fabs(z_axis.dot(x_axis)) > 0.95)
        {
            x_axis = Eigen::Vector3d(0, 1, 0);
        }

        Eigen::Vector3d y_axis = z_axis.cross(x_axis).normalized();
        x_axis = y_axis.cross(z_axis).normalized();

        Eigen::Matrix3d rotation;
        rotation.col(0) = x_axis;
        rotation.col(1) = y_axis;
        rotation.col(2) = z_axis;
        grasp_pose.linear() = rotation;

        geometry_msgs::msg::PoseStamped grasp_msg;
        grasp_msg.header.frame_id = "world";
        grasp_msg.pose.position.x = grasp_pose.translation().x();
        grasp_msg.pose.position.y = grasp_pose.translation().y();
        grasp_msg.pose.position.z = grasp_pose.translation().z();

        Eigen::Quaterniond quat(grasp_pose.rotation());
        grasp_msg.pose.orientation.x = quat.x();
        grasp_msg.pose.orientation.y = quat.y();
        grasp_msg.pose.orientation.z = quat.z();
        grasp_msg.pose.orientation.w = quat.w();

        stage->addPose(grasp_msg);

        // --- Add pose to stage ---
        stage->addPose(grasp_msg);

        Eigen::Isometry3d grasp_frame_transform = Eigen::Isometry3d::Identity();

        Eigen::Matrix3d frame_rotation;
        frame_rotation.col(0) = -Eigen::Vector3d::UnitY(); 
        frame_rotation.col(1) = Eigen::Vector3d::UnitX(); 
        frame_rotation.col(2) = Eigen::Vector3d::UnitZ(); 
        grasp_frame_transform.linear() = frame_rotation;

        grasp_frame_transform.translation().z() = 0.1;

        // Compute IK
        auto wrapper =
            std::make_unique<stages::ComputeIK>("grasp pose IK", std::move(stage));
        wrapper->setMaxIKSolutions(8);
        wrapper->setMinSolutionDistance(1.0);
        wrapper->setIKFrame(grasp_frame_transform, hand_frame_);
        wrapper->properties().configureInitFrom(Stage::PARENT, {"eef", "group"});
        wrapper->properties().configureInitFrom(Stage::INTERFACE, {"target_pose"});
        grasp->insert(std::move(wrapper));
    }
```

Do you think there is a way to combine both approaches. Sample the Position of the object with the GenerateGrapsPose and then fix the orientation with a FixedCartesianPose. I haven't been able to make it work so far however.

rhaschke, 2025-04-07:
In order to use the _current_ object pose, you need to compute the desired pose within the `compute()` method of a stage. Otherwise (as in your `FixedCartesianPoses` example) the pose will be fixed wrt the initial object pose.
The easiest way is probably copying and adapting the `GenerateGraspPose` stage. The `GenerateRandomPose` stage suggested earlier doesn't fit as well, as it doesn't consider the object pose.
As a more structured approach, we could introduce a _relative pose generator_, which can be used in combination with a seed pose to generate various modified poses.

therealbeni, 2025-04-09:
I have now created a new stage that does exactly that. I called it the GenerateRelativePose stage. You can set an Eigen::Vector3d as the approach direction. This direction is then used to create a grasp frame with the y axis aligned to the approach vector. Do you want the files?