moveit/moveit_task_constructor #669: fails to build demo

sussybot5258, 2025-03-20:
i'm on ros2 branch, and tries to build it inside docker, but it says that it fails to find this header file: `ik_clearance_cost_parameters.hpp`
```
#12 648.2 --- stderr: moveit_task_constructor_demo
#12 648.2 /home/container_user/ws_moveit/src/moveit_task_constructor/demo/src/ik_clearance_cost.cpp:11:10: fatal error: moveit_task_constructor_demo/ik_clearance_cost_parameters.hpp: No such file or directory
#12 648.2    11 | #include <moveit_task_constructor_demo/ik_clearance_cost_parameters.hpp>
#12 648.2       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
#12 648.2 compilation terminated.
#12 648.2 gmake[2]: *** [CMakeFiles/moveit_task_constructor_demo_ik_clearance_cost.dir/build.make:76: CMakeFiles/moveit_task_constructor_demo_ik_clearance_cost.dir/src/ik_clearance_cost.cpp.o] Error 1
#12 648.2 gmake[1]: *** [CMakeFiles/Makefile2:257: CMakeFiles/moveit_task_constructor_demo_ik_clearance_cost.dir/all] Error 2
#12 648.2 gmake[1]: *** Waiting for unfinished jobs....
#12 648.2 In file included from /home/container_user/ws_moveit/src/moveit_task_constructor/demo/src/pick_place_task.cpp:38:
#12 648.2 /home/container_user/ws_moveit/src/moveit_task_constructor/demo/include/moveit_task_constructor_demo/pick_place_task.h:60:10: fatal error: moveit_task_constructor_demo/pick_place_demo_parameters.hpp: No such file or directory
#12 648.2    60 | #include <moveit_task_constructor_demo/pick_place_demo_parameters.hpp>
#12 648.2       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
#12 648.2 compilation terminated.
#12 648.2 gmake[2]: *** [CMakeFiles/moveit_task_constructor_demo_pick_place_task.dir/build.make:76: CMakeFiles/moveit_task_constructor_demo_pick_place_task.dir/src/pick_place_task.cpp.o] Error 1
#12 648.2 gmake[1]: *** [CMakeFiles/Makefile2:153: CMakeFiles/moveit_task_constructor_demo_pick_place_task.dir/all] Error 2
#12 648.2 gmake: *** [Makefile:146: all] Error 2
#12 648.2 ---
#12 648.2 Failed   <<< moveit_task_constructor_demo [9.41s, exited with code 2]
```

is there something that i missed?

rhaschke, 2025-03-20:
This file is auto-generated from `demo/src/ik_clearance_cost_parameters.yaml` via [generate_parameter_library](https://github.com/PickNikRobotics/generate_parameter_library). Do you have the latest version of it installed?

sussybot5258, 2025-03-27:
thanks, i've installed both generate_parameter_library and generate_parameter_library_py
and it's working now

closed 2025-03-27 by sussybot5258