moveit/moveit_task_constructor #673: Memory leak when destroying PickPlaceTask objects

Iaroslavrul, 2025-03-31:
### Description
I'm encountering significant memory leaks (~400MB per task) when using MoveIt Task Constructor for pick and place operations. After executing a pick and place task, memory is not properly released.

### Environment

ROS 2 Humble
MoveIt Task Constructor (latest version)
UR5 robot with Robotiq gripper

Issue ### details
When using PickPlaceTask objects for planning and executing pick and place operations, memory is not properly freed after task completion. Each task consumes approximately 400MB of memory which is not released, even after the task objects are explicitly reset and their resources are cleaned up.

[mtc.log](https://github.com/user-attachments/files/19534358/mtc.log)

rhaschke, 2025-03-31:
Could you please provide a minimal code snippet of your MTC usage? Are you using the C++ or Python API?

Iaroslavrul, 2025-03-31:
Hello, @rhaschke 
Sure! This is simplified, "skeletal" code:
// service_node.cpp
void PickPlaceServiceNode::handleExecutePickPlace() {
    auto task = std::make_shared<PickPlaceTask>("pick_place_task");
    if (!task->init(shared_from_this())) {
        RCLCPP_ERROR(get_logger(), "Failed to initialize task");
        return;
    }

    PickPlaceTask::logMemoryStatus(get_logger(), "Before planning: ");

    if (!task->plan(1)) {
        RCLCPP_ERROR(get_logger(), "Planning failed");
        return;
    }

    PickPlaceTask::logMemoryStatus(get_logger(), "Before execution: ");

    if (!task->execute()) {
        RCLCPP_ERROR(get_logger(), "Execution failed");
        return;
    }

    task->releasePluginResources();

    PickPlaceTask::logMemoryStatus(get_logger(), "After cleanup: ");

    std::this_thread::sleep_for(std::chrono::milliseconds(50));
    task.reset();

    PickPlaceTask::logMemoryStatus(get_logger(), "After task reset: ");

    std::this_thread::sleep_for(std::chrono::milliseconds(100));

    PickPlaceTask::logMemoryStatus(get_logger(), "Final: ");
}

// task.cpp
#include <rclcpp/rclcpp.hpp>
#include <moveit/task_constructor/task.h>
#include <moveit/task_constructor/solvers/pipeline_planner.h>
#include <moveit/task_constructor/solvers/joint_interpolation.h>
#include <moveit/task_constructor/stages/fixed_state.h>
#include <atomic>
#include <thread>
#include <memory>

using namespace moveit::task_constructor;

class PickPlaceTask {
public:
    explicit PickPlaceTask(const std::string& task_name);
    ~PickPlaceTask();

    bool init(const rclcpp::Node::SharedPtr& node);
    bool plan(const std::size_t max_solutions);
    bool execute();
    void releasePluginResources();

    static size_t getLiveObjectCount() { return live_objects_; }
    static void logMemoryStatus(const rclcpp::Logger& logger, const std::string& prefix = "") {
        RCLCPP_INFO(logger, "%sMemory status: %zu live objects, %d task counter",
            prefix.c_str(), getLiveObjectCount(), task_counter_.load());
    }

private:
    rclcpp::Node::WeakPtr node_weak_;
    std::string task_name_;
    moveit::task_constructor::TaskPtr task_;
[... 109 lines cut ...]
    if (!task->init(node->shared_from_this())) {
        RCLCPP_ERROR(logger, "Failed to initialize task");
        return;
    }

    PickPlaceTask::logMemoryStatus(logger, "Before planning: ");
    if (!task->plan(1)) {
        RCLCPP_ERROR(logger, "Planning failed");
        return;
    }

    PickPlaceTask::logMemoryStatus(logger, "Before execution: ");
    if (!task->execute()) {
        RCLCPP_ERROR(logger, "Execution failed");
        return;
    }

    task->releasePluginResources();
    PickPlaceTask::logMemoryStatus(logger, "After cleanup: ");

    task.reset();
    PickPlaceTask::logMemoryStatus(logger, "After task reset: ");
}

Would this simplified, "skeletal" code be sufficient for analyzing the memory leak issue?

rhaschke, 2025-04-01:
From this code snippet I cannot recognize a memory leak issue. However, we do have a CI job checking for memory leaks via `asan`. This passes without issues: https://github.com/moveit/moveit_task_constructor/actions/runs/14197818723/job/39777132511

I guess the problem is with your specific code, which probably offers a service to run MTC tasks. If you want me to further analyze the issue, you should 1) provide a compilable and runnable ROS package illustrating the issue and 2) [donate](https://github.com/sponsors/rhaschke). I cannot offer debugging of external code for free anymore.