moveit/moveit_task_constructor #676: How to set isaac cumotion ik planner for pick_place_demo?

lzjie-tchip, 2025-04-11:
Hi,
I want to use isaac cumotion planner in  pick_place_demo,I have successfully configured cumotion in moveit：

[image]

As I run demo.launch.py and run.launch.py,Then the following error appears

`
[pick_place_demo-1] [ERROR] [1744335707.491695277] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491699469] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491703629] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491707789] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491712204] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491716396] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491720556] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491724684] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491728876] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491733100] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491737323] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491741483] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491745675] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491749835] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491753995] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491758155] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491762314] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491766442] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [ERROR] [1744335707.491770698] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm_group'
[pick_place_demo-1] [INFO] [1744335707.495314872] [moveit_task_constructor_demo]: Planning failed
`

I think this is because I deleted the kinematics.yaml in order to use cumotion：

**run.launch.py**

`import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from moveit_configs_utils import MoveItConfigsBuilder
import yaml
import xacro

def generate_launch_description():
    moveit_config = (
        MoveItConfigsBuilder("x3plus", package_name="x3plus_moveit_config")
        .robot_description(
            file_path=os.path.join(
                get_package_share_directory('x3plus_moveit_config'),
                'config',
                'robot_X3plus.urdf.xacro',
            ),
        )
        .robot_description_semantic(
            file_path=os.path.join(
                get_package_share_directory('x3plus_moveit_config'),
                'config',
                'robot_X3plus.srdf',
            ),

        )
        .robot_description_kinematics(file_path='config/kinematics.yaml')
        .trajectory_execution(file_path='config/controllers.yaml')
        .to_moveit_configs()
    )

    # Add cuMotion to list of planning pipelines.
    cumotion_config_file_path = os.path.join(
        get_package_share_directory('x3plus_moveit_config'),
        'config',
        'isaac_ros_cumotion_planning.yaml',
    )
    with open(cumotion_config_file_path) as cumotion_config_file:
        cumotion_config = yaml.safe_load(cumotion_config_file)
    moveit_config.planning_pipelines['planning_pipelines'].insert(
        0, 'isaac_ros_cumotion'
    )
    moveit_config.planning_pipelines['isaac_ros_cumotion'] = cumotion_config
    moveit_config.planning_pipelines['default_planning_pipeline'] = 'isaac_ros_cumotion'

    # Load  ExecuteTaskSolutionCapability so we can execute found solutions in simulation
    move_group_capabilities = {"capabilities": "move_group/ExecuteTaskSolutionCapability"}

    # The workarounds below are based on the following ur_moveit_config and ur_description versions
    # ur_moveit_config: github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/2.2.14
    # ur_description: github.com/UniversalRobots/Universal_Robots_ROS2_Description/tree/2.1.5

    moveit_config.trajectory_execution = {
        'moveit_simple_controller_manager': moveit_config.trajectory_execution
    }
    del moveit_config.trajectory_execution['moveit_simple_controller_manager'][
        'moveit_manage_controllers'
    ]
    moveit_config.trajectory_execution['moveit_manage_controllers'] = True
    moveit_config.trajectory_execution['trajectory_execution'] = {
        'allowed_start_tolerance': 0.01
    }
    moveit_config.trajectory_execution['moveit_controller_manager'] = (
        'moveit_simple_controller_manager/MoveItSimpleControllerManager'
    )

    moveit_config.robot_description_kinematics['robot_description_kinematics'][
        'robot_manipulator'] = moveit_config.robot_description_kinematics[
            'robot_description_kinematics']['/**'][
                'ros__parameters']['robot_description_kinematics']['robot_manipulator']
    del moveit_config.robot_description_kinematics['robot_description_kinematics']['/**']

    moveit_config.joint_limits['robot_description_planning'] = xacro.load_yaml(
        os.path.join(
            get_package_share_directory(
                'x3plus_moveit_config'), 'config', 'joint_limits.yaml',
        )
    )
    moveit_config.moveit_cpp.update({'use_sim_time': True})

    # Add limits from ur_moveit_config joint_limits.yaml to limits from ur_description
    for joint in moveit_config.joint_limits['robot_description_planning']['joint_limits']:
        moveit_config.joint_limits['robot_description_planning']['joint_limits'][joint][
            'has_acceleration_limits'] = True
        moveit_config.joint_limits['robot_description_planning']['joint_limits'][joint][
            'max_acceleration'] = 5.0

    package = "moveit_task_constructor_demo"
    package_shared_path = get_package_share_directory(package)
    node = Node(
        package=package,
        executable=LaunchConfiguration("exe"),
        output="screen",
        parameters=[
            moveit_config.robot_description,
            moveit_config.robot_description_semantic,
            moveit_config.robot_description_kinematics,
            moveit_config.joint_limits,
            moveit_config.planning_pipelines,
            os.path.join(package_shared_path, "config", "panda_config.yaml"),
        ],
    )

    arg = DeclareLaunchArgument(name="exe")
    return LaunchDescription([arg, node])`

How can I use cumotion in pick_place_demo? Or how to set that in cpp code? Thanks in advance

lzjie-tchip, 2025-04-11:
I seem to have set up cumotion successfully by:
`auto sampling_planner = std::make_shared<solvers::PipelinePlanner>(node,"isaac_ros_cumotion");
        sampling_planner->setProperty("goal_joint_tolerance", 1e-5);
        sampling_planner->setPlannerId("cuMotion");
`
There are still errors, though：
`No kinematics solver instantiated for group 'arm_group'`

But the task appears to run, but an error occurs during the grasp pose IK phase：no IK found.

[image]

But the picture shows that the cylinder should be within reach of the robot. How can I solve this problem or identify an accessible location for the robot to place the cylinder? Thanks

devika-dudo, 2025-04-22:
hiii!, I am also facing a similar issue with my custom robot , can you please share if you were able to
 solve the issue