moveit/moveit_task_constructor #678: Unable to visualize tasks in rviz using moveit task constructor

devika-dudo, 2025-04-22:
I was actually facing an error of "no IK found" so I was trying to debug it by visualizing the steps in rviz 

[image]
but cant see them in rviz
```
``devika@devika:~/robo_ws$ ros2 launch hello_moveit_task_constructor run.launch.py exe:=pick_place_demo
[INFO] [launch]: All log files can be found below /home/devika/.ros/log/2025-04-22-17-27-11-212514-devika-68400
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [pick_place_demo-1]: process started with pid [68401]
[pick_place_demo-1] [INFO] [1745323034.891599171] [pick_place_demo]: Initializing task pipeline
[pick_place_demo-1] [INFO] [1745323035.060483465] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.1 seconds
[pick_place_demo-1] [INFO] [1745323035.060682370] [moveit_robot_model.robot_model]: Loading robot model 'urdf_humble_test'...
[pick_place_demo-1] [WARN] [1745323035.060725182] [moveit_robot_model.robot_model]: Skipping virtual joint 'virtual_joint' because its child frame 'base_link' does not match the URDF frame 'world'
[pick_place_demo-1] [INFO] [1745323035.060747671] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[pick_place_demo-1] [WARN] [1745323036.099483236] [moveit_robot_model.robot_model]: Link camera_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[pick_place_demo-1] [INFO] [1745323036.209700584] [pick_place_demo]: OMPL planner created for the arm group
[pick_place_demo-1] [INFO] [1745323036.209855141] [pick_place_demo]: Joint Interpolation planner created for the gripper group
[pick_place_demo-1] [INFO] [1745323036.209899140] [pick_place_demo]: OMPL planner created for the gripper group
[pick_place_demo-1] [INFO] [1745323036.209950053] [pick_place_demo]: Cartesian planner created
[pick_place_demo-1] [INFO] [1745323036.252959136] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[pick_place_demo-1] [INFO] [1745323036.271787055] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.path_tolerance' was not set. Using default value: 0.100000
[pick_place_demo-1] [INFO] [1745323036.271913466] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.resample_dt' was not set. Using default value: 0.100000
[pick_place_demo-1] [INFO] [1745323036.271929180] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.min_angle_change' was not set. Using default value: 0.001000
[pick_place_demo-1] [INFO] [1745323036.272007470] [moveit_ros.fix_workspace_bounds]: Param 'ompl.default_workspace_bounds' was not set. Using default value: 10.000000
[pick_place_demo-1] [INFO] [1745323036.272055590] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_bounds_error' was set to 0.100000
[pick_place_demo-1] [INFO] [1745323036.272073330] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[pick_place_demo-1] [INFO] [1745323036.272108599] [moveit_ros.fix_start_state_collision]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[pick_place_demo-1] [INFO] [1745323036.272153436] [moveit_ros.fix_start_state_collision]: Param 'ompl.jiggle_fraction' was set to 0.050000
[pick_place_demo-1] [INFO] [1745323036.272204489] [moveit_ros.fix_start_state_collision]: Param 'ompl.max_sampling_attempts' was not set. Using default value: 100
[pick_place_demo-1] [INFO] [1745323036.272246882] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[pick_place_demo-1] [INFO] [1745323036.272270279] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Resolve constraint frames to robot links'
[pick_place_demo-1] [INFO] [1745323036.272283548] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[pick_place_demo-1] [INFO] [1745323036.272297028] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[pick_place_demo-1] [INFO] [1745323036.272310297] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[pick_place_demo-1] [INFO] [1745323036.272323218] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[pick_place_demo-1] [INFO] [1745323036.283605610] [pick_place_demo]: Start searching for task solutions
[pick_place_demo-1] [INFO] [1745323038.517427089] [moveit_ros.fix_start_state_bounds]: Starting state is just outside bounds (joint 'joint_2'). Assuming within bounds.
[pick_place_demo-1] [INFO] [1745323038.517504332] [moveit_ros.fix_start_state_bounds]: Starting state is just outside bounds (joint 'joint_3'). Assuming within bounds.
[pick_place_demo-1] [INFO] [1745323038.525362325] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'arm_group' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [WARN] [1745323038.591843002] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint velocity limits are not defined. Using the default 1 rad/s. You can define velocity limits in the URDF or joint_limits.yaml.
[pick_place_demo-1] [WARN] [1745323038.592069144] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[pick_place_demo-1] [INFO] [1745323038.593174015] [moveit.ros_planning.planning_pipeline]: Planning adapters have added states at index positions: [ 0 ]
[pick_place_demo-1] [INFO] [1745323038.598298045] [moveit_ros.fix_start_state_bounds]: Starting state is just outside bounds (joint 'joint_6'). Assuming within bounds.
[pick_place_demo-1] [INFO] [1745323038.599402777] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1745323038.615586969] [moveit.ros_planning.planning_pipeline]: Planning adapters have added states at index positions: [ 0 ]
[pick_place_demo-1] [ERROR] [1745323039.144395536] [pick_place_demo]: Task planning failed with error code: -1
[pick_place_demo-1] [ERROR] [1745323039.144574257] [pick_place_demo]: Failure explanation: Failing stage(s):
[pick_place_demo-1] grasp pose IK (0/30): 0.209440 no IK found
[pick_place_demo-1] 
[pick_place_demo-1] [INFO] [1745323039.144600447] [pick_place_demo]: Detailed stage summary:
[pick_place_demo-1] [INFO] [1745323039.144644236] [pick_place_demo]: pick_place_task: 0 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.144686560] [pick_place_demo]:   applicability test: 1 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.144712820] [pick_place_demo]:     Solution: cost = 0.000, comment = 
[pick_place_demo-1] [INFO] [1745323039.144738102] [pick_place_demo]:     current state: 1 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.144762965] [pick_place_demo]:       Solution: cost = 0.000, comment = 
[pick_place_demo-1] [INFO] [1745323039.144786152] [pick_place_demo]:   Type: Wrapper
[pick_place_demo-1] [INFO] [1745323039.144806335] [pick_place_demo]:   move to safe pose: 1 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.144830640] [pick_place_demo]:     Solution: cost = 0.652, comment = 
[pick_place_demo-1] [INFO] [1745323039.144855433] [pick_place_demo]:   move to safe hand pose: 1 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.144878690] [pick_place_demo]:     Solution: cost = 0.500, comment = 
[pick_place_demo-1] [INFO] [1745323039.144902505] [pick_place_demo]:   open gripper: 1 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.144926041] [pick_place_demo]:     Solution: cost = 19.500, comment = 
[pick_place_demo-1] [INFO] [1745323039.144953000] [pick_place_demo]:   move to pick: 0 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.144977025] [pick_place_demo]:   pick object: 0 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.144996231] [pick_place_demo]:     approach object: 0 solutions, 0 failures
[... cut ...]
[pick_place_demo-1] [INFO] [1745323039.146999874] [pick_place_demo]:     retreat after place: 0 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.147022153] [pick_place_demo]:   Type: Serial Container
[pick_place_demo-1] [INFO] [1745323039.147045480] [pick_place_demo]:   move home: 0 solutions, 0 failures
[pick_place_demo-1] [INFO] [1745323039.147067340] [pick_place_demo]: Type: Serial Container
[pick_place_demo-1] [INFO] [1745323039.147149053] [pick_place_demo]: Planning failed

THis is the log when i am trying to run the pick_and_place ``

```

rhaschke, 2025-04-22:
I am not aware of a package named `hello_moveit_task_constructor`. In any case, you need to choose the correct solution topic. Using the pick&place demo from `moveit_task_constructor_demo`, the topic is named `/solution`:
```
ros2 launch moveit_task_constructor_demo demo.launch.py
ros2 launch moveit_task_constructor_demo run.launch.py exe:=pick_place_demo
```

devika-dudo, 2025-04-22:
[This is my workspace](https://github.com/devika-dudo/manipulatorr) --- hello_moveit_task_contructor is the package for my custom robot , also the 

[image] solution topic is publishing --- inorder to see the tasks how should i set the solution topic in rviz?

rhaschke, 2025-04-22:
Just set the field `Task Solution Topic` to `/solution`. rviz will also display failures (not only valid solutions).
You need to ensure that your MTC process keeps running beyond the planning call to allow rviz to communicate with it.
Just insert a sleep at the end of your program.

devika-dudo, 2025-04-22:
There's no topics available when i click the field near task solution topic , also noticed that my /solution topic isn't publishing any data when i echoed it. 
```

``bool PickPlaceTask::plan(const std::size_t max_solutions) {
  RCLCPP_INFO(LOGGER, "Start searching for task solutions");
  try {
    moveit::core::MoveItErrorCode error_code = task_->plan(max_solutions);
    if (error_code == moveit::core::MoveItErrorCode::SUCCESS) {
      RCLCPP_INFO(LOGGER, "Task planning completed successfully. Found %zu solutions", task_->numSolutions());
      
      if (!task_->solutions().empty()) {
        RCLCPP_INFO(LOGGER, "Publishing solution for visualization");
        task_->introspection().publishSolution(*task_->solutions().front());
      }
      // Print detailed stage summary
      RCLCPP_INFO(LOGGER, "Detailed stage summary:");
      printStageDetails(task_->stages(), 0);  // Pass 0 as the initial indentation level
      return true;
    } else {
      RCLCPP_ERROR(LOGGER, "Task planning failed with error code: %d", error_code.val);
      
      // Print detailed failure explanation
      std::ostringstream explanation;
      task_->explainFailure(explanation);
      RCLCPP_ERROR(LOGGER, "Failure explanation: %s", explanation.str().c_str());
      
      // Print detailed stage summary even if planning failed
      RCLCPP_INFO(LOGGER, "Detailed stage summary:");
      printStageDetails(task_->stages(), 0);  // Pass 0 as the initial indentation level
      return false;
    }
  } catch (const std::exception& ex) {
    RCLCPP_ERROR(LOGGER, "Exception during planning: %s", ex.what());
    return false;
  }
}
```
this is part of my pick_place_task.cpp that deals with /solution 
Also my  MTC process is running beyond the planning call by adding this to my pick_place_demo.cpp
`spinning_thread.join();`
but i am still not able to visulaize

rhaschke, 2025-04-22:
> There's no topics available when i click the field near task solution topic.

Looks like the dropdown is not populated in ROS2. Enter `/solution` manually.

> Also noticed that my `/solution` topic isn't publishing any data when i echoed it.

`publishSolution()` is only called on success. So, that's not a surprise.
However, other topics are derived from the `/solution` topic, namely `/description` and `/statistics`. These should be active during planning.

devika-dudo, 2025-04-23:
Thanks for your reply, now I am able to see the stages when I manually typed the name of the topic, but I cant see the #,cost,comment when i click each of them --- am I missing something?

[image]

rhaschke, 2025-04-23:
You minimized the solution property widget to width zero. Just use the [Qt splitter handle](https://www.youtube.com/watch?v=Hbh9BMtgE50&ab_channel=MacDigia) to show the widget again (or recreate the whole MTC display).