moveit/moveit_task_constructor #681: How to fix  [Connect]: Deviation in joint in a connect stage

lesurJ, 2025-04-29:
Hello,

I encounter an issue with a Connect stage that prevents a motion plan to be found when not using the `<plugin>mock_components/GenericSystem</plugin>` in my robot ros2_control.xacro file (e.g `<plugin>topic_based_ros2_control/TopicBasedSystem</plugin>`).

My code is based on the [pick place task example](https://github.com/moveit/moveit_task_constructor/blob/humble/demo/src/pick_place_task.cpp). This code works fine when using the default `<plugin>mock_components/GenericSystem</plugin>` in the ros2_control file (perfect system). The motion plan is found very quickly and properly executed.

However, when changing the ros2_control.xacro file to connect to IsaacSim, my console is filled with the following.

```bash
[station_mover-6] [INFO] [1745914767.580166114] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'MP3_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[station_mover-6] [INFO] [1745914767.688871034] [Connect]: Deviation in joint MP3_finger_joint: [0] != [0.0297]
[station_mover-6] [INFO] [1745914767.690307839] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'MP3_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[station_mover-6] [INFO] [1745914767.764132468] [Connect]: Deviation in joint MP3_finger_joint: [0] != [0.0297]
[station_mover-6] [INFO] [1745914767.765025067] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'MP3_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[station_mover-6] [INFO] [1745914767.838515628] [Connect]: Deviation in joint MP3_finger_joint: [0] != [0.0297]
[station_mover-6] [INFO] [1745914767.840090475] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'MP3_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[station_mover-6] [INFO] [1745914767.937191546] [Connect]: Deviation in joint MP3_finger_joint: [0] != [0.0297]
[station_mover-6] [INFO] [1745914767.938711842] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'MP3_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[station_mover-6] [INFO] [1745914768.026757974] [Connect]: Deviation in joint MP3_finger_joint: [0] != [0.0297]
[station_mover-6] [INFO] [1745914768.027871199] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'MP3_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[station_mover-6] [INFO] [1745914768.100738602] [Connect]: Deviation in joint MP3_finger_joint: [0] != [0.0297]
[station_mover-6] [INFO] [1745914768.101132869] [station_mover]: Planning failed
```

The planning output is:
```bash
[station_mover-6]   0  - ←   0 →   -  0 / pick_place_task
[station_mover-6]     1  - ←   1 →   -  0 / current state
[station_mover-6]     -  0 →   1 →   -  1 / move home
[station_mover-6]     -  1 →   0 ←  95  - / move to pick
[station_mover-6]    95  - ←  95 →   - 95 / pick object
[station_mover-6]      95  - ←  95 ←   0  - / approach object
[station_mover-6]       0  - ←  95 →   -  0 / grasp pose IK
[station_mover-6]        25  - ←  25 →   - 25 / generate grasp pose
[station_mover-6]       -  0 →  95 →   -  0 / allow collision (hand,object)
[station_mover-6]       -  0 →  95 →   -  0 / close hand
[station_mover-6]       -  0 →  95 →   -  0 / attach object
[station_mover-6]       -  0 →  95 →   -  0 / allow collision (object,support)
[station_mover-6]       -  0 →  95 →   -  0 / lift object
[station_mover-6]       -  0 →  95 →   - 95 / forbid collision (object,surface)
[station_mover-6] Failing stage(s):
```

The MP3_finger_joint value is indeed 0.0297 according to isaac sim but the moveit package somehow expect it to be zero. This makes the planning fail as no solution can be found for the connect stage (see the "move to pick" stage in the following picture):

[image]

I've looked into the documentation without finding the way to handle this. 

Is there a way to relax the joint constraint for a Connect stage?

Setup:
- Ubuntu 22.04
- ROS2 Humble, 
- MoveIt binary install
- MTC source install (humble branch)
- Custom moveit config (made with setup assistant) for a yaskawa robot

rhaschke, 2025-04-29:
Compare the final states of the stages `move home` and `approach object` in both situations.
I guess, in the first condition (`GenericSystem`) the finger joint is identical in both states (probably equal to 0.0297).
In the second condition (`TopicBasedSystem`) the states are different (0 vs. 0.0297).

This means, that the connect stage not only should plan for the arm motion but also for the gripper motion:

https://github.com/moveit/moveit_task_constructor/blob/5a0059dc83e60307797787b0f2d594e0ca30db48/demo/src/pick_place_task.cpp#L223-L225

lesurJ, 2025-04-29:
You are right, and the proposed solution is working.
Thanks a lot!

closed 2025-04-29 by lesurJ