moveit/moveit_task_constructor #682: how to set orientation constraint in connect stage?

amoyalang, 2025-05-22:
When I perform the grasping and placing actions, I hope that the clamped object will not tilt during the process of moving to the placing position. Therefore, I added path constraints in the` connet `stage, which led to the planning always failing. I need to consult everyone, so I used the panda robot to simulate my process of setting path constraints, and the result was also a planning failure.

I use [this demo](https://github.com/moveit/moveit2_tutorials/blob/main/doc/tutorials/pick_and_place_with_moveit_task_constructor/src/mtc_node.cpp)  
reproduct problem. I only added orientation constraints in the connect phase. The following is my connect stage code and my understanding of direction constraints.
```c++
{
        moveit_msgs::msg::OrientationConstraint &c =
        path_constraints.orientation_constraints[0];
        c.link_name = "panda_hand";
        c.header.frame_id = "panda_link0";
        c.orientation.w = 1.0;

        //When grasping object,X axis is upright, 
        //so the X axis could be rotated,it is tolerancec omplete freedom,ohter tolerances shouble be constrained
        c.absolute_x_axis_tolerance = M_PI;
        c.absolute_y_axis_tolerance = M_PI / 10;
        c.absolute_z_axis_tolerance = M_PI / 10;
        c.parameterization = moveit_msgs::msg::OrientationConstraint::XYZ_EULER_ANGLES;
        c.weight = 1.0;
}
```
When an object is being clamped, the X-axis of panda_hand is upward. I don't want the X-axis of panda_hand to deviate during the movement process. Therefore, I think it is not allowed to rotate significantly around the y and z axes, but it can rotate freely around the X-axis. My code is written based on this idea.

[image]
This has led to its inability to plan successfully.
Why is this?

rhaschke, 2025-05-26:
> it always plan failed

In your code, you specify the target orientation for `panda_hand`, which probably isn't in the identity orientation when grasping the object. Thus, the constraint is not satisfied initially. This should be stated in the commandline logs as well.

rhaschke, 2025-08-26:
Planning with constraints is - in principle - possible with MoveIt2. However, it's not possible to specify the constraint w.r.t. an attached object. You need to specify it w.r.t. a link.

closed 2025-08-26 by rhaschke

nvspranathi, 2025-09-05:
> Have a look at the cmdline log. It typically states useful error messages.

Hello @rhaschke , since then I've been trying to work on moveit with path constraints. My application is to hold a glass of water without tilting it throughout the path. I'm trying it on a doosan robot. I've tried many ways but seeing the issue- unable to solve planning problem. It works fine for some points but for a little bit far points, the planning fails. Can you point out any mistakes that might lead to this error or tell me any other working alternatives for my application. I'd be very grateful to you if I could get a solution for this.