moveit/moveit_task_constructor #683: Error al ejecutar solución MTC en RViz: “does not have any controllers specified for trajectory execution”

IBRAHIM1006, 2025-05-22:
— Estoy usando ROS 2 (Jazzy Jalisco) y MoveIt Task Constructor (MTC) para planificar y ejecutar un flujo pick-and-place sobre un robot MyCobot en RViz.
— Tras completar el planning en MTC y pulsar “Exec” en RViz, no veo ningún movimiento en el simulador ni en hardware.
En el terminal aparece esta advertencia:

[move_group-1] [WARN] [1747935087.235500977] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '0' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-1] [INFO] [1747935087.235519537] [moveit_task_constructor_visualization.execute_task_solution]: Executing TaskSolution
[move_group-1] [INFO] [1747935087.235584392] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list

ROS 2 distro: Jazzy Jalisco
Sistema operativo: Ubuntu 24.04 en WSL2 
simulador: Gazebo Harmonic

este es un fragmento del código del nodo mtc_node  Author: Addison Sears-Collins:  

  auto stage_state_current = std::make_unique<mtc::stages::CurrentState>("current state");
  current_state_ptr = stage_state_current.get();
  task.add(std::move(stage_state_current));

  auto stage_open_gripper =
    std::make_unique<mtc::stages::MoveTo>("open gripper", interpolation_planner);
  stage_open_gripper->setGroup(gripper_group_name);
  stage_open_gripper->setGoal(gripper_open_pose);
  stage_open_gripper->properties().set("trajectory_execution_info",
                      mtc::TrajectoryExecutionInfo().set__controller_names(controller_names));
  task.add(std::move(stage_open_gripper));

En un inicio funcionaba todo bien pero luego actualice ROS2 y tuve esos errores.
— ¿Qué estoy pasando por alto para que MTC no asigne los controllers a la trayectoria?
— ¿Hay algún parámetro adicional o método en MTC que indique explícitamente qué controlador usar al ejecutar la solución?

rhaschke, 2025-05-23:
Please translate your question to English.

IBRAHIM1006, 2025-05-23:
—I’m using ROS 2 (Jazzy Jalisco) and MoveIt Task Constructor (MTC) to plan and execute a pick-and-place workflow on a MyCobot robot in RViz.
—After completing the planning in MTC and clicking “Exec” in RViz, I don’t see any movement in either the simulator or the hardware.

The following warning appears in the terminal:
[mtc_node-1] [INFO] [1748010206.779034873] [mtc_node]: Execution skipped as per configuration
[mtc_node-1] [INFO] [1748010206.779042873] [mtc_node]: Task execution completed. Keeping node alive for visualization. Press Ctrl+C to exit.
[move_group-1] [WARN] [1748010253.697379027] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '0' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-1] [WARN] [1748010253.698347327] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '0' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-1] [WARN] [1748010253.698511327] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '0' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-1] [WARN] [1748010253.698643027] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '0' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-1] [INFO] [1748010253.698659127] [moveit_task_constructor_visualization.execute_task_solution]: Executing TaskSolution
[move_group-1] [INFO] [1748010253.698720827] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1748010253.698823328] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1748010253.699084028] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1748010253.699571328] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1748010253.699617828] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-1] [INFO] [1748010253.700096028] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-1] [INFO] [1748010253.700160128] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1748010253.700196728] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-1] Stack trace (most recent call last) in thread 133029:
[move_group-1] #6    Object "", at 0xffffffffffffffff, in
[move_group-1] #5    Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 78, in clone3 [0x7f9ca1ef3c3b]
[move_group-1] #4    Source "./nptl/pthread_create.c", line 447, in start_thread [0x7f9ca1e66aa3]
[move_group-1] #3    Source "../../../../../src/libstdc++-v3/src/c++11/thread.cc", line 104, in execute_native_thread_routine [0x7f9ca20f8db3]
[move_group-1] #2    Object "/opt/ros/jazzy/lib/libmoveit_trajectory_execution_manager.so.2.12.3", at 0x7f9ca1afc429, in trajectory_execution_manager::TrajectoryExecutionManager::executeThread(std::function<void (moveit_controller_manager::ExecutionStatus const&)> const&, std::function<void (unsigned long)> const&, bool)
[move_group-1] #1    Object "/opt/ros/jazzy/lib/libmoveit_plan_execution.so.2.12.3", at 0x7f9ca1d0b6c4, in plan_execution::PlanExecution::successfulTrajectorySegmentExecution(plan_execution::ExecutableMotionPlan const&, unsigned long)
[move_group-1] #0    Object "/home/ibrahim/ros2_ws/install/moveit_task_constructor_capabilities/lib/libmoveit_task_constructor_capabilities.so", at 0x7f9c682efd96, in move_group::ExecuteTaskSolutionCapability::constructMotionPlan(moveit_task_constructor_msgs::msg::Solution_<std::allocator<void> > const&, plan_execution::ExecutableMotionPlan&, std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_task_constructor_msgs::action::ExecuteTaskSolution> > const&)::{lambda(plan_execution::ExecutableMotionPlan const*)#1}::operator()(plan_execution::ExecutableMotionPlan const*) const [clone .isra.0]
[move_group-1] Segmentation fault (Signal sent by the kernel [(nil)])
[ERROR] [move_group-1]: process has died [pid 132443, exit code -11, cmd '/opt/ros/jazzy/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_4a0mng60 --params-file /tmp/launch_params_hn86b6xg --params-file /tmp/launch_params_r1hj1s3e --params-file /tmp/launch_params_4r4lnw2v'].

##ROS 2 distro: Jazzy Jalisco
Operating system: Ubuntu 24.04 en WSL2
simulator: Gazebo Harmonic

##This is a snippet of the node's code:

auto stage_state_current = std::make_uniquemtc::stages::CurrentState("current state");
current_state_ptr = stage_state_current.get();
task.add(std::move(stage_state_current));

auto stage_open_gripper =
std::make_uniquemtc::stages::MoveTo("open gripper", interpolation_planner);
stage_open_gripper->setGroup(gripper_group_name);
stage_open_gripper->setGoal(gripper_open_pose);
stage_open_gripper->properties().set("trajectory_execution_info",
[... cut ...]
task.add(std::move(stage_open_gripper));

##These are my controllers.
    arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    gripper_action_controller:
      type: position_controllers/GripperActionController

— What am I missing that causes MTC not to assign controllers to the trajectory?
— Is there any additional parameter or method in MTC to explicitly specify which controller to use when executing the solution?
— What exactly is stage '0'?
I would appreciate your response.

rhaschke, 2025-05-23:
The message `The trajectory of stage '0' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.` is just a warning. This shouldn't cause a segfault.
Try to drop the call to `stage_open_gripper->properties().set("trajectory_execution_info", ...)`

IBRAHIM1006, 2025-05-31:
"I've already tried the suggested steps. The thing is, my project in ROS 2 Jazzy was working fine until May 15th, 2025, when my system completely shut down. Then, on May 16th, it started showing that error without me making any changes. What could have happened?