moveit/moveit_task_constructor #694: Enable task execution in moveit_task_constructor_demo

pkkulhari, 2025-06-13:
The task execution is disabled by default in `ros2 launch moveit_task_constructor_demo run.launch.py exe:=pick_place_demo`.

How to enable it, I tried to pass `execute:=true` param, but it does not work.

ros version: ros2 humble

rhaschke, 2025-06-13:
Check wether [existing issues](https://github.com/moveit/moveit_task_constructor/issues?q=is%3Aissue%20exec) are related!
If not, provide the full log output.

pkkulhari, 2025-06-13:
log says `Execution disabled` at the end.

```
ros2 launch moveit_task_constructor_demo run.launch.py exe:=pick_place_demo
[INFO] [launch]: All log files can be found below /home/praveen/.ros/log/2025-06-13-16-58-49-664855-pop-os-185653
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [pick_place_demo-1]: process started with pid [185654]
[pick_place_demo-1] [INFO] [1749814130.891591722] [moveit_task_constructor_demo]: Initializing task pipeline
[pick_place_demo-1] [INFO] [1749814130.900153251] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00144507 seconds
[pick_place_demo-1] [INFO] [1749814130.900172290] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[pick_place_demo-1] [INFO] [1749814130.916483565] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[pick_place_demo-1] [INFO] [1749814130.918031921] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.path_tolerance' was not set. Using default value: 0.100000
[pick_place_demo-1] [INFO] [1749814130.918041208] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.resample_dt' was not set. Using default value: 0.100000
[pick_place_demo-1] [INFO] [1749814130.918043512] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.min_angle_change' was not set. Using default value: 0.001000
[pick_place_demo-1] [INFO] [1749814130.918057279] [moveit_ros.fix_workspace_bounds]: Param 'ompl.default_workspace_bounds' was not set. Using default value: 10.000000
[pick_place_demo-1] [INFO] [1749814130.918064189] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_bounds_error' was set to 0.100000
[pick_place_demo-1] [INFO] [1749814130.918066360] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[pick_place_demo-1] [INFO] [1749814130.918071272] [moveit_ros.fix_start_state_collision]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[pick_place_demo-1] [INFO] [1749814130.918073249] [moveit_ros.fix_start_state_collision]: Param 'ompl.jiggle_fraction' was not set. Using default value: 0.020000
[pick_place_demo-1] [INFO] [1749814130.918078714] [moveit_ros.fix_start_state_collision]: Param 'ompl.max_sampling_attempts' was not set. Using default value: 100
[pick_place_demo-1] [INFO] [1749814130.918084037] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[pick_place_demo-1] [INFO] [1749814130.918087215] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Resolve constraint frames to robot links'
[pick_place_demo-1] [INFO] [1749814130.918088595] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[pick_place_demo-1] [INFO] [1749814130.918089987] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[pick_place_demo-1] [INFO] [1749814130.918091241] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[pick_place_demo-1] [INFO] [1749814130.918092613] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[pick_place_demo-1] [INFO] [1749814130.919583332] [moveit_task_constructor_demo]: Start searching for task solutions
[pick_place_demo-1] [INFO] [1749814130.934515838] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [WARN] [1749814130.938917780] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[pick_place_demo-1] [INFO] [1749814130.984272127] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1749814131.008076039] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1749814131.027390688] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1749814131.054671216] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1749814131.073124312] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1749814131.090490753] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[... cut ...] t when the planner is constructed.
[pick_place_demo-1] [INFO] [1749814133.110151950] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1749814133.135397893] [moveit_task_constructor_demo]: Planning succeded
[pick_place_demo-1] [INFO] [1749814133.135416503] [moveit_task_constructor_demo]: Execution disabled

```

rhaschke, 2025-10-22:
The `execute` parameter is defined here:
https://github.com/moveit/moveit_task_constructor/blob/44d6e1a1c21f1a302f98b0352047075a2c0a404a/demo/src/pick_place_demo_parameters.yaml#L2-L4

closed 2025-10-22 by rhaschke